Bodmer/TJpg_Decoder #7: TJpg_Decoder library compatible with BGR screens?

marlonhiram12, 2020-04-22:
I'm doing a project in the university like this one:
https://youtu.be/TfuEI1-YrsA

But for a different application. I did everything and I can see the video streaming of the esp32 cam in the TFT screen thanks to your library.

I realized that my video is a little bit darker. I made a few test and when I change the video in the camera webserver to a red filter, in the TFT the video applied a blue filter. My TFT works with BGR formar, not normal RGB.

I don't know if there's an option in this library to convert the colors to BGR.

Bodmer, 2020-04-22:
Displays as you have found vary and have different color order. The [TFT_eSPI library has an option](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L63-L64) to change this.

The jpeg library itself should be compatible with the jpeg standard and therefore output a standard color order. However there are processor endianess issues which impact the byte order, so this may be the problem. In this case there is code to [handle endainess differences](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/tjpgd.c#L1664-L1682), just use the [swapBytes() function](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/TJpg_Decoder.h#L100), try both true and false as function parameters.

What is your hardware setup?

marlonhiram12, 2020-04-22:
> Displays as you have found vary and have different color order. The [TFT_eSPI library has an option](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L63-L64) to change this.

I already made this and doesn't work and nothing changed.

> just use the [swapBytes() function](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/TJpg_Decoder.h#L100), try both true and false as function parameters.

I actually have the swapByte in true in my void setup()

`void setup() {`
`// put your setup code here, to run once:`
`Serial.begin(115200);`
`delay(1000);`
`  tft.begin();`
`  tft.setRotation(3);`
`  tft.setTextColor(0xFFFF, 0x0000);`
`  tft.fillScreen(TFT_RED);`
**`  tft.setSwapBytes(true); // We need to swap the colour bytes (endianess)`**

If i declare false swapBytes the tft screen shows the video of the esp32cam like a rainbow...  

> In this case there is code to [handle endainess differences](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/tjpgd.c#L1664-L1682),

    if (jd->swap)
    {
      do {
        w = (*s++ & 0xF8) << 8;     // RRRRR-----------
        w |= (*s++ & 0xFC) << 3;    // -----GGGGGG-----
        w |= *s++ >> 3;             // -----------BBBBB
        *d++ = (w << 8) | (w >> 8); // Swap bytes
      }   while (--n);
    }

When I declare true setSwapBytes, you mean like the first byte (0000 0000 1111 1111) shifts to the  second byte? (1111 1111 0000 0000) and viceversa?

Bodmer, 2020-04-22:
What is your hardware setup. I assume now it is not an ILI9341 TFT...

marlonhiram12, 2020-04-22:
https://microcontrollerslab.com/2-4-tft-lcd-touch-screen-module/

I'm using this TFT LCD screen 8 bits parallel

Bodmer, 2020-04-22:
The web page is does not appear to be a module specification. It mentions the ILI9341 but then recommends the driver library for the SPFD5408 controller which is not supported by TFT_eSPI.

One way to fint out the controller is to run the [MCUFRIEND_kbv library](https://github.com/prenticedavid/MCUFRIEND_kbv)  [LCD_ID_readreg example](https://github.com/prenticedavid/MCUFRIEND_kbv/tree/master/examples/LCD_ID_readreg) which will report the screen controller type.

Bodmer, 2020-05-16:
Do you have it working yet?

marlonhiram12, 2020-05-17:
It didn't work with the 8 bit parallel TFT, I bought other TFT with SPI pins (MOSI, MISO, SCKL, CD, DC, RST). With the SPI TFT all work very well. 
Thank you anyways....

closed 2020-05-17 by marlonhiram12