Bodmer/TJpg_Decoder #8: Getting individual RGB values in tft_output callback

fcapano, 2020-04-27:
I'm trying to use this decoder on an ESP32 to output images to an RGB LED matrix (using the SmartMatrix library).

I can't make sense of the format of the data passed in the callback.
In this example, I'm passing in an image whose first three pixels are: pure red, pure green and pure blue. I then print in binary the contents of the first three memory locations:
`
static bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap)
{
    for (int o=0; o<w*h; o++) {
        uint16_t v = bitmap[o];
        for (int i = 0; i < 16; i++) {
            printf("%d", (v & 0x8000) >> 15);
            v <<= 1;
        }
        printf("\n");
    }
`
The output:
`
0001111100000000
1110000000000111
0000000011111000
`
What is the encoding here? Perhaps I've messed something up since red seems to be encoded with five bits and green with six.
Is there a way to fetch the individual RGB values for each pixel?

Bodmer, 2020-04-27:
Hello, as you have surmised the output is in 16 bit "565" RGB format which is good for small TFT displays. If you want 888 RGB fotmat then [change this flag](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/tjpgd.h#L10) to 0.

closed 2020-05-03 by Bodmer

token47, 2020-10-31:
Hi! I have exactly the same problem. It's difficult to change that flag to zero when you imported the lib from arduino framework (you can't change that file). You could just define that in your sketch before the include, but it will fail with "JD_FORMAT redefined" because you did not put it inside and ifdef to check if it's already defined. Can you dot it in the next version? Thanks!

token47, 2020-10-31:
Actually, you will also need to change this:

typedef bool (*SketchCallback)(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t *data);

It will not let you use different type on the bitmap data. You could have two types (for 16 and 24 bits) and two definitions of the same function one for each.