Bodmer/TJpg_Decoder #11: Any demand for an example showing Web_jpg but with https images?

witnessmenow, 2020-07-20:
Hey,

Thanks for the library, I've been using with my spotify API library to display album art and my daughter loves it!

[image]

As part of this work I needed to fetch the image off a https source and I was wondering is it worth me submitting a PR with an example of it? it probably could even be included in the original example.

Gist for what I worked on is here, but I would clean it up for a PR:
https://gist.github.com/witnessmenow/063dcd4b4407aa51afb4f6a0a442b5af

Bodmer, 2020-07-27:
An example would be welcome. Ideally it should support both ESP32 and ESP8266, I am not sure what you are using but code for both using https can be [found here](https://github.com/Bodmer/OpenWeather/blob/master/OpenWeather.cpp).

Bodmer, 2020-07-28:
I have just looked at your gist link. I can see the code is for a bespoke display. To be included the example sketch would need to be converted to a commonly available TFT. Also all potentially revenue earning links would need to be removed.

closed 2020-07-28 by Bodmer