Bodmer/TJpg_Decoder #14: Web_Jpg.ino Not Working

tonyc770, 2020-10-28:
I have used your demo program to try out downloading and displaying a JPG image with the ESP8266 and ILI9488 TFT.
 When I run the example code I get the following output from the serial monitor,

.....WiFi Connected...
For test only, removing file

Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
/EagleEye.jpg                         37755 bytes
/Mouse.jpg                            27808 bytes
=================================================

Downloading /F35.jpg from http://i.imgur.com/OnW2qOO.jpg
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... **code: 301**
118 ms to download

Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
/EagleEye.jpg                         37755 bytes
/Mouse.jpg                            27808 bytes
**/F35.jpg                                  0 bytes**
=================================================

1 ms to draw to TFT

As you can see there is an HTTP Code 301, and the file size of the F35.jpg file is 0 bytes.

Any suggestions?

Thanks,
Tony

leandrobviana, 2020-11-04:
@tonyc770 i was having the same problem with this same example, i changed the URL to HTTPS ([https://i.imgur.com/OnW2qOO.jpg](https://i.imgur.com/OnW2qOO.jpg)) and it is now working for me. 
I am using a Wemos D1 R32 and a ILI9481 Parallel.

Bodmer, 2020-11-05:
Code, 301 means the file has moved. It looks like Imgur has changed, it no longer accepts "http" GET requests, it now only accept "https" requests, I.e. secure ones. So the 301 code means it has moved to https. I will update the example, thanks for reporting this.

Bodmer, 2020-11-05:
Change does not work for ESP8266

Bodmer, 2020-11-05:
I have uploaded the fix for the ESP8266

commit f7b010bb (2020-11-05): Fix #14

Changed http to https for GET request.
See #14.
commit 19cc900c (2020-11-05): Fix #14 for ESP8266

Fix #14
closed 2020-11-05 by Bodmer