Bodmer/TJpg_Decoder #15: Web Fetch Causes ESP32 to Reboot

tonyc770, 2020-11-25:
I have noticed that every so often as the Web Fetch is getting a JPG file from the web (I download and display a weather radar map) every 15 minutes, the ESP32 will reboot because of a WatchDog timer issue.

Here is the output from the Serial Monitor:
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200
E (1425376) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (1425376) task_wdt:  - IDLE0 (CPU 0)
E (1425376) task_wdt: Tasks currently running:
E (1425376) task_wdt: CPU 0: esp_timer
E (1425376) task_wdt: CPU 1: loopTask
E (1425376) task_wdt: Aborting.
abort() was called at PC 0x400d224b on core 0

Backtrace: 0x4008c6b4:0x3ffbe160 0x4008c8e5:0x3ffbe180 0x400d224b:0x3ffbe1a0 0x400842c9:0x3ffbe1c0 0x4000bfed:0x3ffb7400 0x40089f69:0x3ffb7410 0x4008bc73:0x3ffb7430 0x4008c2f8:0x3ffb7450 0x4008456c:0x3ffb7470 0x400845a1:0x3ffb7490 0x40084a65:0x3ffb74b0 0x4000beaf:0x3ffb74d0 0x4010ae44:0x3ffb74f0 0x4010aed3:0x3ffb7510 0x4010af41:0x3ffb7530 0x4010817b:0x3ffb7550 0x4010a4f3:0x3ffb7570 0x4010ad0d:0x3ffb75b0 0x401530d6:0x3ffb75e0 0x401530e5:0x3ffb7610 0x40153132:0x3ffb7630 0x40145fb5:0x3ffb7660 0x40147d3b:0x3ffb7680 0x40147dad:0x3ffb76a0 0x4014842a:0x3ffb76c0 0x401413a1:0x3ffb76e0 0x4013f359:0x3ffb7700 0x4016b033:0x3ffb7720 0x40139945:0x3ffb7740 0x40139ad9:0x3ffb7890 0x40139b72:0x3ffb78d0 0x400d1c24:0x3ffb78f0 0x400d1c8b:0x3ffb7910 0x40088dfd:0x3ffb7930

Rebooting...

Any ideas?

Thanks,

Bodmer, 2020-11-25:
It looks like a watchdog time-out. I exect there is a long wait for something and the software is looping. Try finding out where this loop is and adding a yield(); to restart the watchdog.

tonyc770, 2020-11-27:
I added an extra yield() function to the main loop() and the function that retrieves the JPG image. That seems to fix the issue with the Watchdog timeout.

Thanks...

closed 2020-11-28 by Bodmer