Bodmer/TJpg_Decoder #17: Feature Request

dallen98, 2021-01-13:
I,m trying to do a 180 x 360 pixel neomatrix, to this end i,m using an esp32 attached to ili9341 tft to experiment, I want 3 videos each 7seconds long running at 25-30 fps, jpegs (or giffs) could this Tjpg decoder do that or perhaps you can point me in the right direction, all help much appreciated thanks dave

Bodmer, 2021-01-13:
The ESP32 does not have the processing power to decode the images that fast. You might achieve around 10 fps.

This GIF library has may be worth a try. I supplied one of the examples for use with the TFT_eSPI library.

Bodmer, 2021-01-13:
[AnimatedGI library link](https://github.com/bitbank2/AnimatedGIF).

Bodmer, 2021-01-13:
I tested the ESP32_Dual_Core_Flash_Jpg example and with a 240x320 panda test image each frame takes 63ms, so ~16fps. The render time is proportional to the image size and interface speed to the display. To get 25-30fps the image would need to be about160x240 pixels.  The Dual core example uses the ESP32's second processor to decode the image and the other to render the image to the display.

The JPEGDEC library here will decode jpeg images  a bit faster (maybe 1.5x depending on the image), it uses more RAM lookup tables for the decoding but should run fine on an ESP32.

dallen98, 2021-01-13:
in the JPEGDecoder/examples/Other libraries/SPIFFS_Jpeg/Data/  there are 2 static jpegs a babboon etc do you have a animated jpeg that I could play in the jpegdecoder ?

Bodmer, 2021-01-13:
The library does not handle animated or progressive jpegs. You would have to have a set of images and load them individually in sequence.

dallen98, 2021-01-13:
is there a dual core flash animated gif example anywhere or in the pipeline ?? Thanks

dallen98, 2021-01-13:
quite a good video where 3 minutes in he has created a prgramme that converts animated gifs to .h files
https://www.youtube.com/watch?v=-h9Vm0Ow_Is

Bodmer, 2021-01-14:
Thanks for the link, it is interesting and useful.

The video poster is creating "raw" images where each pixel is represented by a 16 bit value, this means the complex maths involved in decoding a jpeg image can be avoided and thus the frame rate increases.  I ran a test on the "walking man" animation with no loop delays and it ran at 297 fps. This is close to the maximum theoretical speed but this is a much smaller image that you would require. Raw image data occupies a lot of space, 180x360 at 30fps for 7s would require more than 20Mbytes of data to be stored, Jpegs might only occupy 2Mbytes but then you have the extra decode time.

I think you ought to start experimenting with the Neomatrix and see what data rates you can get.

closed 2021-01-14 by Bodmer

dallen98, 2021-01-14:
I,m told that whatever works on a tft ili9341 should work on neomatrix, I have lowered my expectations on this to a 180x180 matrix (hoping I can use 2 esp32 in synchroniastion) 

thinking of ways forward a animated gif using dual cores like youve achieved with jpegs would be brill :)

dallen98, 2021-01-14:
...or a tft-espi version of jpegdec https://hackaday.com/2020/08/26/new-arduino-jpeg-library-focuses-on-speed/

Bodmer, 2021-01-14:
I have not used Neomatrix modules, but I think they typically come in 8x8 or 32x32 modules, this means 180x180 appears to be a rather an odd choice. It is good to start with a setup where the aims can be met easily rather than waste a lot of time trying to get the performance you need.

The Hackaday link refers to the JPEGDEC library I muentioned in post 4 above but forgot to add a link, the author is the same as the AninatedGIF library I mentioned also.

dallen98, 2021-01-14:
the neomatrix modules clones i bought from china came in 16x16, these are ws2812b leds also available in strip form 3metres long 180 leds per strip which i can wire to my own matrix requirments to achieve a 180x180 or 180x360 matrix hence i,m working in muliples of 180.
The gif to .h converter linked above takes forever to convert the smallest file a 50kb gif ends up 3 hours later as 2mb, the other thought I had was using .dat file... created in xlights software... https://github.com/ShaunPrice/FSEQLib

Bodmer, 2021-01-14:
You ought to consider more than just the software. The power demand is going to be high, 180 x 180 pixels at 30mA each is 972 Amps. You will need multiple distributed power supples and a very good grounding scheme!

Hope it works out for you. Since these are more project related questions rather than a library issue I will close.