Bodmer/TJpg_Decoder #19: Get raw array directly

beckmx, 2021-02-08:
Hello, I was wondering if the codec can be used only as a decoder-only, I noticed that the setCallback is called after the extraction of the file from spiffs:
`TJpgDec.drawFsJpg(0, 0, "file.jpg");`

But was wondering if it was possible to only receive the decoded array to store it somewhere else in an inline way:
`uint16_t bitmap = TJpgDec.decodeJpg(0, 0, myConcatenation);`

Bodmer, 2021-02-08:
The Tjpg_Decoder library is a "decoder" only library, so it cannot encode and image into jpg format.

The examples receive the decoded "Minimum Coding Units" (MCU) from the decoder and render each tile to the screen.  Typically an MCU is a 16x16 image tile but they can be, for example, 8x8 or 8x16. All these sizes are accomodated by the library.

The example functions could be adapted but if you want a raster scan image then you would need to buffer and store complete MCU rows. [This function](https://github.com/Bodmer/TJpg_Decoder/blob/master/examples/Flash_Jpg/Flash_Jpg.ino#L18-L34) receives the MCU blocks and renders them to the screen one at a time, the function is passed a pointer to the memory block containing the image tile, and the width and height of the tile.

Note that the right hand and bottom edge times may be of a different size as they are cropped, so you must extract the clipped image from the image block. This is what [this code does](https://github.com/Bodmer/JPEGDecoder/blob/master/examples/Other%20libraries/SPIFFS_Jpeg/JPEG_functions.ino#L89-L104) (yes it is a different jpg library but that is what you will need to do for clipped MCUs.)

closed 2021-02-08 by Bodmer

beckmx, 2021-02-08:
Based on your response, I think you thought I was asking about encoding, I was actually asking if there is a direct method to get the decoded image (just like it is) but without the callback, I am thinking in using the result array in lvgl

Bodmer, 2021-02-08:
No, but your comment "decoder-only" implied there was encoding capability. It is already a decoder-only library.  The image is decoded in small blocks called MCUs, and these are fed to the call-back function tft_output. This is the way it works to decode the image incrementally and it does not work in any other way. If you wish to save a raster scan copy of the image then you will need to adjust that function (rename if you like) and buffer a line of MCUs. Typical CPUs like the ESP32 do not have enough memory to store images over approximately 200x200 pixels and, when rendering to the TFT, there is not need to store a complete image. However as you surmise this is possible as detailed above.

I advise you [read this to understand how jpegs are decoded](https://en.wikipedia.org/wiki/JPEG#Block_splitting).

Bodmer, 2021-02-08:
This is a[ simpler less technical explanation](https://www.impulseadventure.com/photo/jpeg-minimum-coded-unit.html).