Bodmer/TJpg_Decoder #21: Getting exception(29) sometimes when saving an image from the web using Web_Fetch functions

bmarcu13, 2021-03-02:
Sometimes when I try to download multiple images my ESP8266 resets itself (before trying to download one of the images) giving this error:

```
22:31:49.969 -> Exception (29):
22:31:49.969 -> epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
22:31:50.062 -> 
22:31:50.062 -> >>>stack>>>
22:31:50.062 -> 
22:31:50.062 -> ctx: sys
22:31:50.110 -> sp: 3fffec10 end: 3fffffb0 offset: 0190
22:31:50.157 -> 3fffeda0:  00000006 3ffedfd8 00000002 40100448  
22:31:50.157 -> 3fffedb0:  4025bcfb 00000030 00000014 4025bc90  
22:31:50.250 -> 3fffedc0:  00000002 4025bc37 00000002 4025ad90  
22:31:50.296 -> 3fffedd0:  4025adb9 3fffee80 3ffedfd8 00000016  
22:31:50.343 -> 3fffede0:  40258820 3fffee80 3ffede84 3ffed888  
22:31:50.389 -> 3fffedf0:  3ffebbe0 3fffee80 3fffee80 4023d7a7  
22:31:50.436 -> 3fffee00:  74756c50 3fff006f 00000000 4023825f  
22:31:50.485 -> 3fffee10:  3fff0f74 00000000 00000000 3fff0df4  
22:31:50.530 -> 3fffee20:  40254b7d 00001539 00001539 40100813  
22:31:50.624 -> 3fffee30:  ffffffb1 3ffed998 3ffebbf0 3ffedfd8  
22:31:50.673 -> 3fffee40:  3ffece18 00000011 00000000 40259517  
22:31:50.718 -> 3fffee50:  00000000 3fff11c4 ffffffb1 00000000  
22:31:50.765 -> 3fffee60:  00000000 3ffedfd8 00000000 00000010  
22:31:50.812 -> 3fffee70:  3fff0f74 3fff1070 0bcadf2a 402371e2  
22:31:50.859 -> 3fffee80:  00000000 00110b0b 00320104 00000044  
22:31:50.906 -> 3fffee90:  3ffebc04 000000ce 3ffebc18 3ffebbf8  
22:31:50.953 -> 3fffeea0:  00000000 3ffebc04 3ffebc0b 3ffebc21  
22:31:51.000 -> 3fffeeb0:  00000000 00000000 3ffebc27 3ffebcb8  
22:31:51.047 -> 3fffeec0:  3ffebc41 3ffebc5d 00000000 3ffebc8f  
22:31:51.143 -> 3fffeed0:  00000000 00000000 00000011 00000000  
22:31:51.189 -> 3fffeee0:  3fff1954 40258f8a 3ffece18 3fff11c4  
22:31:51.237 -> 3fffeef0:  00000000 3ffedfd8 3ffece18 3ffebbe0  
22:31:51.283 -> 3fffef00:  3ffebbe0 000000f6 00000000 00000011  
22:31:51.331 -> 3fffef10:  00000000 3ffebbea 402626e7 3ffece18  
22:31:51.376 -> 3fffef20:  3ffebbd4 3fffdcc0 3ffe94c0 3ffe94c0  
22:31:51.423 -> 3fffef30:  00000080 3ffece18 00000000 3ffe8540  
22:31:51.518 -> 3fffef40:  40261fa7 3fffdab0 00000000 40220448  
22:31:51.518 -> 3fffef50:  3ffe94c0 40000f49 3fffdab0 40000f49  
22:31:51.612 -> 3fffef60:  40000e19 00000005 00085854 00000000  
22:31:51.612 -> 3fffef70:  00000000 aa55aa55 00000065 40104a5d  
22:31:51.707 -> 3fffef80:  40104a63 00085854 00000000 ffffffff  
22:31:51.754 -> 3fffef90:  4010000d ffffffff ffffffff ffffffff  
22:31:51.801 -> 3fffefa0:  40260e80 3fffef3c 40260e39 3ffffc38  
22:31:51.848 -> 3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
22:31:51.895 -> 3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
22:31:51.987 -> 3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
22:31:51.987 -> 3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.081 -> 3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.081 -> 3ffff000:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.175 -> 3ffff010:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.221 -> 3ffff020:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.268 -> 3ffff030:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.314 -> 3ffff040:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.361 -> 3ffff050:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.455 -> 3ffff060:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.455 -> 3ffff070:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.549 -> 3ffff080:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.549 -> 3ffff090:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.642 -> 3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.693 -> 3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.738 -> 3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.783 -> 3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.829 -> 3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.922 -> 3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
22:31:52.922 -> 3ffff100:  feefeffe feefeffe feefeffe feefeffe  
22:31:53.016 -> 3ffff110:  feefeffe feefeffe feefeffe feefeffe  
22:31:53.016 -> 3ffff120:  feefeffe feefeffe feefeffe feefeffe  
22:31:53.108 -> 3ffff130:  feefeffe feefeffe feefeffe feefeffe  
22:31:53.155 -> 3ffff140:  feefeffe feefeffe feefeffe feefeffe  
22:31:53.203 -> 3ffff150:  feefeffe feefeffe feefeffe feefeffe  
[... 51 lines cut ...]
22:31:55.922 -> 3ffff490:  3c2c974f c49e2f24 7e2a94f1 cc0e4c88  
22:31:55.968 -> 3ffff4a0:  75f750bc 4a37d88b 634a1070 ae154e0f  
22:31:56.018 -> 3ffff4b0:  4ef2625e ca2277ab 5d6210d4 96e84f99  
22:31:56.064 -> 3ffff4c0:  762c5fa1 bf5636ae 37f90223 09428b55  
22:31:56.155 -> 3ffff4d0:  827adada 7ec80d60 e6655a59 ebb848b2  
22:31:56.155 -> 3ffff4e0:  150d094d 828d7fe0 1c59e006 4b45f3c8  
22:31:56.250 -> 3ffff4f0:  92bc672e 51490b4d 3fd5b5fd 95cecef3  
22:31:56.250 -> 3ffff500:  80000000 00000000 00000000 00000000  
22:31:56.340 -> 3ffff510:  00000000 00000000 00000000 00000300  
22:31:56.388 -> 3ffff520:  a3b99eed fc1f88b6 fde99321 d771eaba  
22:31:56.436 -> 3ffff530:  60e10988 aa834a9a 2bf15e5b 16bd2ea3  
22:31:56.483 -> 3ffff540:  010b9587 3069d9e3 8fe50cfb 00000080  
22:31:56.529 -> 3ffff550:  03030000 010000e2 eb7bdd69 0000ee1c  
22:31:56.624 -> 3ffff560:  5cecea77 6726a45a 478e2bcc cdd1086f  
22:31:56.624 -> 3ffff570:  ec1f0000 1e7a5621 a8c02bc0 5acca9cc  
22:31:56.719 -> 3ffff580:  30c0acc0 2fc02cc0 afc023c0 adc0aec0  
22:31:56.719 -> 3ffff590:  28c009c0 27c024c0 14c02dc0 13c00ac0  
22:31:56.813 -> 3ffff5a0:  32c025c0 31c02ec0 2ac004c0 29c026c0  
22:31:56.813 -> 3ffff5b0:  0f009c00 0ec005c0 9dc0a0c0 9dc09cc0  
22:31:56.909 -> 3ffff5c0:  3d002f00 a1003c00 12c003c0 35c008c0  
22:31:56.955 -> 3ffff5d0:  57b90fac 0d2676b0 76225f84 a4aa973a  
22:31:57.001 -> 3ffff5e0:  4bfd0a7a 85987046 f4f7a46b 57e73b33  
22:31:57.047 -> 3ffff5f0:  d44864de 395ea73e ccf9f5aa 1e486b71  
22:31:57.095 -> 3ffff600:  e786dfef c321d054 0924eb7f 5afeb169
```

Bodmer, 2021-03-27:
The library itself is a decoder and has no web/network download capability. Typically images are decoded and saved if SPIFFS or LITTLEFS before decoding as the ESP8266 has limited RAM.

Bodmer, 2021-04-24:
I assume this has been resolved - closing.

closed 2021-04-24 by Bodmer