Bodmer/TJpg_Decoder #22: Problem while rendering JPEG Image.

viveky1794, 2021-03-11:
I am doing a project where I need to store my jpeg file into SPIFFS for later use.
I am  using Arduino IDE. So I am  putting 2 JPEG images into "data"  folder.
Then Updating after going "Tool>ESP32 sketch upload" section.

> Arduino IDE
> ESP32 Dev it
> Rpi TFT, 480x320, ili9486

@Problem :
One Image name " LOGO.jpg " Is rendering. But Second image namely "StartImage.jpg" is not rendering. Can anyone suggest the possible reason and solution.

Koxx3, 2021-03-25:
did you find any solution ? I have the same trouble (panda sample displaying ... my image isn't. no progressive jpg, nothing special)

Bodmer, 2021-03-27:
I tried the second image above and can confirm that it will not be decoded.  I then openend it in MS Paint, saved it again and it worked fine. Here is that image:

[TJPG_Issue_22p.zip](https://github.com/Bodmer/TJpg_Decoder/files/6214926/TJPG_Issue_22p.zip)

So there must be something incompatible in the format, maybe it contains a "thumb nail image?

Bodmer, 2021-03-27:
I tried the [JPEGDecoder library](https://github.com/Bodmer/JPEGDecoder) which uses a different decode engine and it reports "Jpeg file format not supported!", so this means it is encoded in an unsupported format.

The JPEGDecoder  library did load the image version posted above.

I ran analysis on the original image and it is in progressive format:

[image]

closed 2021-03-27 by Bodmer