Bodmer/TJpg_Decoder #29: Error compiling for board AI Thinker ESP32-CAM

jokikill, 2021-11-12:
Hello,

I have an issue with the TJpg_Decoder on my esp32-cam code.

I tryed to create an IR vision on a tft screen, the ir is op but for the tft I have an error "...\Arduino\libraries\TJpg_Decoder-master\src/TJpg_Decoder.h:31:30: fatal error: LITTLEFS.h: No such file or directory for board AI Thinker ESP32-CAM".

For add a tft screen on my esp32-cam I followed this webpage : https://www.survivingwithandroid.com/esp32-cam-tft-display-picture-st7735/

Can you help me?

alextief, 2021-11-18:
Hi Jokikill,
Im on the same Project like you. Try adding the fs.h library at the beginning (#include <fs.h>) or/and littlefs.h. 

But i cant compile because Bodmers TJpg_Decoder references BYTECLIP in some files and I cant get it to work...
"undefined reference to `BYTECLIP'"

Bodmer, 2021-11-18:
@jokikill Which version of the Arduino ESP32 board package are you using? The latest is 2.0.1.

@jokikill BYTECLIP issue has been resolved.

alextief, 2021-11-18:
> @jokikill Which version of the Arduino ESP32 board package are you using? The latest is 2.0.1.
> 
> @jokikill BYTECLIP issue has been resolved.

Ill will try it now if it works.
Anyways thank you very much for your work!!!

Bodmer, 2021-11-18:
Looks like there was a change in 2.0.0  "Rename LITTLEFS to LittleFS to match ESP8266"....

https://github.com/espressif/arduino-esp32/commit/c8a4010f

Bodmer, 2021-11-18:
I have renamed header file reference so the library will now only work on ESP32 with board package 2.0.0 or later.

closed 2021-11-18 by Bodmer

ajacobs-github, 2021-11-23:
When compiling for board ESP32S2 Dev Module I get the error undefined reference to `BYTECLIP' in tjpgd.c.  Arduino ESP32 version 2.0.1.  Board is sparkfun ESP32-S2 Thing Plus and display is Adafruit 2.8" TFT LCD shield with cap touch and micro SD

Bodmer, 2021-11-24:
@ajacobs-github  This has been fixed in version 1.0.2 and should be available in the Arduino Libary manager soon.

jokikill, 2021-11-26:
> Hi Jokikill, Im on the same Project like you. Try adding the fs.h library at the beginning (#include <fs.h>) or/and littlefs.h.
> 
> But i cant compile because Bodmers TJpg_Decoder references BYTECLIP in some files and I cant get it to work... "undefined reference to `BYTECLIP'"

I'm on the last update for it.

With the last update of Jpg_Decoder, I have always this error : 

In file included from ...\CamToIli9341\CamToIli9341.ino:21:0:
...\Arduino\libraries\TJpg_Decoder-master\src/TJpg_Decoder.h:23:26: fatal error: LittleFS.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board AI Thinker ESP32-CAM.

jokikill, 2021-11-26:
I've switched on this tutorial for an ILI9341 https://blog.xuite.net/iamleon/blog/589421027

pipi61, 2022-11-14:
try change
static uint8_t BYTECLIP (int val)
to
uint8_t BYTECLIP (int val)
in tjpgd.c

Bodmer, 2022-11-15:
Interestingly the this simple change degrades rendering performance from 84ms to 99ms on the ESP32 for the Flash_Jpg example! So it is not an ideal change from that perspective for users that do not get the error (which seems to be specific to PlatformIO compilations, not the Arduino IDE ones).

hiibrarahmad, 2023-01-06:
[image]

> When compiling for board ESP32S2 Dev Module I get the error undefined reference to `BYTECLIP' in tjpgd.c. Arduino ESP32 version 2.0.1. Board is sparkfun ESP32-S2 Thing Plus and display is Adafruit 2.8" TFT LCD shield with cap touch and micro SD

try to add link like this it solve the issue in platform.io
[image]