Bodmer/TJpg_Decoder #30: Error with ESP32 LittleFS

gmbo, 2021-11-25:
When I use the library with an ESP32 I get an error message.
LittleFS.h not found.

The error is probably only on case sensitive drives, because the libraries differ by the notation.
ESP8266 uses LittleFS.h and
ESP32 uses LITTLEFS.h
A change in TJpg_Decoder.h can help.
`
  #if defined (ESP8266) || defined (ESP32)
    #include <pgmspace.h>
    #include <FS.h>
    #ifdef ESP32
    #define SPIFFS LITTLEFS
    #include <LITTLEFS.h>
    #else
    #include <LittleFS.h>
    #endif

    #define TJPGD_LOAD_FFS
  #endif
`
After that the example Animated_dial with settings for ESP32 can be compiled without errors.

Wenn ich die Library mit einem ESP32 benutze bekomme ich eine Fehlermeldung.
LittleFS.h nicht gefunden.

The text below was translated into English with DEEPL.com
Der untenstehende Text wurde mit DEEPL.com ins Englische übersetzt

Der Fehler ist warscheinlich nur auf Case Sensitiv Laufwerken, da die Libraries sich durch die Schreibweise unterscheiden.
ESP8266  nutzt  LittleFS.h und
ESP32 nutzt LITTLEFS.h
Eine Änderung in TJpg_Decoder.h schafft Abhilfe.
`
  #if defined (ESP8266) || defined (ESP32)
    #include <pgmspace.h>
    #include <FS.h>
    #ifdef ESP32
    #define SPIFFS LITTLEFS
    #include <LITTLEFS.h>
    #else
    #include <LittleFS.h>
    #endif

    #define TJPGD_LOAD_FFS
  #endif
`
Danach läßt sich das Beispiel Animated_dial mit Einstellungen für den ESP32 fehlerfrei übersetzen.

Bodmer, 2021-11-26:
The latest 2.0.x  ESP32 board package has LittleFS.h

This package update was made for compatibility with the ESP8266. 

If you do not with to updatethe ESP32board package then you will need to use the previous version of TJpg_Decoder.

gmbo, 2021-11-26:
Thanks for the info.
I still had the LittleFS_esp32 library from lorol installed, since the code was not included with ESP32 earlier.
I will remove the library and test it.

closed 2021-11-27 by Bodmer