Bodmer/TJpg_Decoder #31: 可以获取到图片但是无法显示！

shumei12358, 2022-01-23:
我是一个新手，我尝试使用ESP32开发板与ILI9341使用Web_jpg显示图片，我发现以jpg为结尾的链接可以正常显示，但是以没有jpg结尾的图片可以下载到SPIFFS中，占有正常的内存大小，但是不可以显示，代码在运行时似乎直接跳过了。
`Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
  M81.jpg                               17074 bytes
=================================================

2 ms to draw to TFT
`

Bodmer, 2022-01-25:
Translated using Google translate:
" I'm a newbie, I try to use ESP32 development board with ILI9341 to display pictures using Web_jpg, I found that links ending with jpg can be displayed normally, but pictures ending without jpg can be downloaded to SPIFFS, occupying normal memory size, but No display, the code seems to be skipped directly at runtime. "

Bodmer, 2022-01-25:
I have run the SPIFFS_Web_Jpg example and the picture is drawn. This is the output on the Serial Monitor window:
```
 Testing TJpg_Decoder library

Initialisation done.
.....

Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
=================================================

Downloading /M81.jpg from https://i.imgur.com/C77RWcq.jpg
[HTTP] begin...
[HTTP] GET...
[HTTP] GET... code: 200

[HTTP] connection closed or file end.
2635 ms to download

Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
  M81.jpg                                7857 bytes
=================================================

82 ms to draw to TFT
```

Bodmer, 2022-02-16:
I assume this has been resolved.

closed 2022-02-16 by Bodmer