Bodmer/TJpg_Decoder #32: Jpegdecoder with TFT_eSPI Not working on WT32-SCO1

videobuff, 2022-01-25:
I just got myself this nifty screen + ESP32 called WT32-SCO1.
I 'd like to use it with your sprite technology from TFT_eSPI, but it does no work, wheras with an ILI9488 it works.
Als the samples from the TFT_eSPI library all work, but the samples from the Jpegdecoder do not work. I used sample All_LittleFS.

Further testing showed me this error

```
Testing TJpg_Decoder library
./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
E (873) esp_littlefs: mount failed,  (-84)
E (875) esp_littlefs: Failed to initialize LittleFS
LittleFS initialisation failed!
TFT_eSPI settings are user_select201.
The board selected is is an ESP32 Wrover.

```
Your help to solve this issue is greatly appreciated.

Bodmer, 2022-01-25:
The error sugests that a Flash partition for LittleFS has not been setup in the IDE. After it is set upload the data to LittleFS then upload the sketch. This will make sure the LittleFS partition is available and initialised.

videobuff, 2022-01-26:
You were right. Uploading pics solved the issue. Now I only have to solve my own sketch which runs fine on a ili9488 but shows a white screen on the WT-32. But that’s a different story. Many thanks for your mighty library.

closed 2022-01-26 by videobuff

Bodmer, 2022-01-26:
Glad you have it sorted.

Maybe, run the Read_User_Setup sketch and check the serial monitor output to see if your setup file for the WT-32 is being used by the compiler.