Bodmer/TJpg_Decoder #34: JPG Decoder not correctly decoding. height is always 0

moefear85, 2022-02-04:
I've tried several images from the web, as well as frames captured directly with the esp32-cam module, directly in jpeg format (no conversion). All images have dimensions 320x240. Yet TJpg_Decoder always returns 320 for the width correctly, but always returns 0 for the height. The images are always noise, with general features recognizable that resemble the original image.

Bodmer, 2022-02-05:
It sounds like a byte swap problem for the rendering. Try putting 
tft.setSwapBytes(true); // Also try false
In your sketch setup() .

Does a sketch example work OK?

Provide an example image.

I  am working away so it will be 2 weeks before I can investigate.

Bodmer, 2022-02-16:
I assume this has been resolved. I am not able to help without further information.

closed 2022-02-16 by Bodmer

moefear85, 2022-02-22:
Hi Sorry it took so long to respond... I should have mentioned earlier that I already tried setSwapBytes() both with true and false, as well as so many other possibilities that I've exhausted. Is really noone else experiencing anything similar?

No offense, I really appreciate your software, especially tft_espi (very fast, only one that supports parallel, I've even taken first steps to expose it to micropython). But given that nothing worked, I eventually moved to JPEGDEC.

Bodmer, 2022-02-23:
Post an example image that fails.

Bodmer, 2022-03-03:
I have upgraded the library master to use the [latest decoder engine](http://elm-chan.org/fsw/tjpgd/00index.html). However the decoding process has not changed significantly. Also check your compiler optimisation settings are not having an effect, I tend to use Os as the optimisation level.

boroncsokb, 2023-03-29:
Hi @Bodmer,

I am also having the same issue as described above. Everything worked great until I moved from Platformio to esp-idf (v4.4, using c++), but now I can not get it to work properly. The TFT_eSPI library still works, I can fill the screen with different colors, I can draw shapes on it without any problem, and also the TFT_Flash_Bitmap example works as intended, so I assume, that the communication with the screen is not the root cause.

However as I try to run the Flash_Jpg example from the TJpg_Decoder library (only switching the setup-loop functions to app_main for it to work in esp-idf), the image does not appear properly (see attached photo) 
On the serial output, it also says: "Width = 240, height = 0"
I have already tried setting the SwapBypes to true and false, but I get the same noisy image every time.

Do you have an idea, what could go wrong?

Thank you!
[image]