Bodmer/TJpg_Decoder #35: Image is not rendered properly.

MokhammadDaniyal, 2022-02-16:
Hello! Thank you for the library!

I am running the SPIFFS_Web_Jpg example, just as it is. For some reason the images are not fully rendered. Only about 20% of the image is rendered. I am running the code on ESP8266 with ILI9341 display.

Checking the image size downloaded is 
`/F35.jpg                              10935 bytes
/M81.jpg                               7857 bytes
`
Which seems to be correct.

Any suggestion on what I can do to solve this issue?

Displaying images with another TFT library works fine, so display should not be the problem here.

Thanks!

Bodmer, 2022-02-16:
This seems to be related to an issu ewith SPIFFS on the ESP8266, the sketch runs fine on an ESP32 but not on an ESP8266.

I have tested with LittleFS and it is working so it is not related to the web fetch code.

Can you download the latest library from Github and try the LittleFS example. Since SPIFFS is not deprecated on the ESP8266 it may be a good time to move over to LittleFS.

Bodmer, 2022-02-16:
The issue occurs with version 3.0.x of the Arduino ESP8266 oard package. Rolling back to 2.7.4 fixes the problem.

Since this is an issue with the board package I cannot fix in TFT_eSPI.  I see there is an issue raised on the board package so hopefully the board package team will fix this.

closed 2022-02-16 by Bodmer

MokhammadDaniyal, 2022-02-18:
You are Right! LittleFS works! Have not tested with SPIFFS and 2.7.4.

Thank you very much!!

MokhammadDaniyal, 2022-02-19:
Random question: Have you ever connected it with a MQTT server? I am working on a project where NodeMCU will recieve a Imgur link from MQTT connection and then download image using the example from that link and display on the screen.
I am using this library https://github.com/Jeffin28/AWS-IOT-CORE-ESP8266 because I have setup AWS MQTT server. So the libraries work separately but when I combine them into one project i get exception at line `int httpCode = http.GET();` in `Web_fetch`. Maybe you have had some experience and have some suggestions?

Bodmer, 2022-02-19:
It is difficult to debug code without full visibility. You may find that the problem is not where the exception is generated. Try closing down the AWS connection before connecting to imgur, because there may be a client configuration mismatch between the two clients.