Bodmer/TJpg_Decoder #37: only a red screen comes up on my esp32-s2 board

hannescam, 2022-03-20:
it detects the file with All_SPIFFS but only a red screen and it says the image has a resolution of 0x0
please help
thanks

Bodmer, 2022-03-20:
Does the unmodified example show this behaviour or is the example modified? Make sure plenty of space is allocated for SPIFFS.

Which ESP32 board package version are you using?

SPIFFS is deprecated so maybe try LittleFS which is also faster.

hannescam, 2022-03-20:
i have no modification made just the raw example
the pakage is ESP32S2 Dev Module
with my homemade esp32-s2 board
and the scematic is in pdf here send
i bridged the FE1.1 that means only the cp2104 has a connection

Bodmer, 2022-03-20:
I have tested the sketch with a ESP32-S2 and ILI9341 display and it is working.

This is the serial output:
```
 Testing TJpg_Decoder library

Initialisation done.
/Baboon40.jpg 320x480 1:2 185 ms
/panda.jpg 240x320 1:1 110 ms
/tiger.jpg 240x320 1:1 189 ms
```
Post your serial output.

Also try the Flash_Jpg example as that does not need SPIFFS to be configured and no data files have to be uploaded.

This is the serial output I get with that sketch:
```
 Testing TJpg_Decoder library
Width = 240, height = 320
99 ms
```
Post your output to that sketch as well.

hannescam, 2022-03-20:
it is working with littlefs and the flash_jpeg 
but with littlefs it only showed a red display and this output:` Testing TJpg_Decoder library

Initialisation done.
1.jpg
/1.jpg 0x0 1:1 30 ms
`
and with the tiger it was working all fine the pictures are attached
i experimented with the first one but never showed any diference
[image]
[image]
[image]
[image]

[image]
[image]
[image]
[image]
[image]
d

Bodmer, 2022-03-20:
Ah! So you are modifying the example with you own files!

Not sure what the 1 and 2 are but the links just give me errors.

hannescam, 2022-03-21:
i left the original files also on there and they worked flawlessly

hannescam, 2022-03-21:
here are the pics: [link](https://github.com/hannescam/pics-for-bodmer)
1 and 2 are these from above and the rest 3 - 9 are all others

Bodmer, 2022-03-22:
I had a look at image 1.jpg and it is in 8 bit format which the decoder does not support. Jpegs must be 24 bit. Here is one converted and it loads fine.

To convert I loaded it into MS Paint and then just did a saveas:
[1x.zip](https://github.com/Bodmer/TJpg_Decoder/files/8320333/1x.zip)

closed 2022-03-22 by Bodmer

Bodmer, 2022-03-22:
I have updated the ReamdMe to make this clearer. Thanks for reporting this.

hannescam, 2022-03-22:
thankyou i had a txt to png converter script set for greyscale and i uploaded te png to a converter website and put the jpg file in the spiffs erlier and now littlefs