Bodmer/TJpg_Decoder #38: Animated_dial Compliation error - ESP32 / ILI9341

Amitabh-wq, 2022-05-03:
Hi, I am trying to test the example called 'Animated_dial' but geting the IDE error:

I:\Users\UserJ\Documents\Arduino\libraries\TJpg_Decoder\src/TJpg_Decoder.h:23:26: fatal error: LittleFS.h: No such file or directory

I have installed all the libraries required. Any toughts? Thanks

Bodmer, 2022-05-03:
Load the latest master jpeg library and try that.

Amitabh-wq, 2022-05-03:
> Load the latest master jpeg library and try that.

Hi Bodmer, I am using the TJpg_Decoder latest version 1.0.3 but it has the above compilation error.

Bodmer, 2022-05-03:
I checked with board package 2.0.2 (latest) and it runs fine and LittleFS.h is found.

I suspect you are using an old 1.0.x board package as those version did not provide LittleFS support.

Amitabh-wq, 2022-05-03:
I did the following, I edited the header 'TJpg_Decoder.h' (from the TJpg_Decoder library) I renamed the call of library from <LittleFS.h> to <LITTLEFS.h>, as the LITTLEFS library is in capital letters. I use ESP32 boards V 1.0.6 which is the most stable for my projects. The compilation went without errors, but the graph has problems, the pointer is erasing the numbers, see the photo below:
[image]

Amitabh-wq, 2022-05-03:
In addition, before I also tested ESP Boards V 2.0.2 and this same error occurred.

Bodmer, 2022-05-04:
The display SDO line must be connected. Have you specified the read frequency in the setup file?
#define SPI_READ_FREQUENCY  10000000

Amitabh-wq, 2022-05-04:
Hi Bodmer, I followed your instructions and it worked. Only the numbers in the center are overlapping, I don't know if this is normal. See below

https://user-images.githubusercontent.com/89787514/166835163-2129088c-a6c5-4313-82e5-4917404a16fa.mp4

.

Bodmer, 2022-05-04:
The numbers problem has been fixed in the latest master library on Github.

Amitabh-wq, 2022-05-05:
I am using the latest.
[image]

Bodmer, 2022-05-05:
Latest is 2.4.61 and is only available from Github here a the moment:
https://github.com/Bodmer/TFT_eSPI

Amitabh-wq, 2022-05-05:
> Latest is 2.4.61 and is only available from Github here a the moment: https://github.com/Bodmer/TFT_eSPI

I have installed it manually and replaced with my configurated files. Now it is working fine. Thank you.

Bodmer, 2022-05-05:
Great, I will be releasing the new version soon.

IAmOrion, 2022-07-23:
I'm curious, where are you getting 2.0.2 package from? I came here with the same "LIttleFS" not found problem.  I say @Amitabh-wq comment to use uppercase LITTLEFS and that has worked for me.  As far as I know I have the correct board url > https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json and there's no 2.x.x version?

[image]

Bodmer, 2022-08-02:
Currently I am using the development branch link:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json