Bodmer/TJpg_Decoder #40: TJpgDec.getJpgSize

joseph-vedadi, 2022-05-26:
I do not know how to convert the image to binary 
I used the web-tool provided for diff image the size is zero ... 

```
TJpgDec.getJpgSize(&w, &h, img, sizeof(img));
```
what is the specification of the image, and why it is zero ?

Bodmer, 2022-05-26:
I now use the tool and[ instructions detailed here](https://github.com/Bodmer/TFT_eSPI/discussions/1843).

Jpeg file arrays should start with the marker values: 0xFF,0xD8,0xFF,

closed 2022-05-26 by Bodmer

joseph-vedadi, 2022-05-26:
@Bodmer Thank you but I still have the same issue 

[image]
[image]

Bodmer, 2022-05-26:
A PNG format file cannot be decoded with a jpeg decoder. Convert the png to a jpeg first with an image handling program like GIMP or MS Paint. Alternatively use a PNG decoder library.