Bodmer/TJpg_Decoder #43: TJpg_Decoder.h seems conflict with ESP32-CAM face detecttion function fmt2rgb888()

lioujj, 2022-06-25:
When I include TJpg_Decoder.h and fd_forward.h (library for ESP-32-CAM face detection), ESP32 will crash.
Always run error at fmt2rgb888() function in header file img_converters.h.
If I don't include TJpg_Decoder.h, the function fmt2rgb888() runs with no error.

Thanks.

lioujj, 2022-06-25:
Include TJpg_Decoder.h and fd_forward.h same time, ESP32 will crash when run at fmt2rgb888()
[image]

But if only include fd_forward.h, ESP32 runs correctly, the other code is the same.
[image]

Bodmer, 2022-06-26:
Do the TJpg_Decoder library examples work OK?

It is rather odd that the TJpg_Decoder library compiles at all with ESP32 1.0.6 board package since v 1.0.5 requires LittleFS....

The crash may be down to lack of memory resources, the library requires at least 3500 bytes of RAM for a working buffer.

Bodmer, 2022-06-26:
According to the documentation the mt2rgb888 will convert the campera image from JPEG format to the RGB888 format. This means the face detection sketch already includes a jpeg decoder, thus it is not clear why you are including another jpeg decoder.  It is possible both libraries use the same decoder engine so this may be the root cause of the crash.

closed 2022-06-26 by Bodmer

lioujj, 2022-06-27:
Thanks!
I included the TJpg_Decoder for display the camera image on TFT instantly, It works perfectly. I want to show the image on TFT and detect face at the same time, but it crashed.