Bodmer/TJpg_Decoder #47: Unable to read JPEG files from SDcard.

vick3394, 2022-09-01:


vick3394, 2022-09-01:
Dear Mr Bodmer,
Thank you and your colleagues for all the efforts you put in to give us a good experience with TFT screens.
I have used the 2.8" TFT with ILI9341 ESP32 with no issues. I have also used the program written by XTronical which allows random reading of SD card of JPEG images stored 240x 320 with no issues.

I recently gotten a 4" TFT from Aliexpress. It indicated that I should be using with driver ILI9488. 
I have connected everything as per the 2.8" TFT. 
I was able to run the example(TFT_meter) after changing the setup to driver ILI9488.

When I run the program by Xtroni
![unnamed](https://user-images.githubusercontent.c
[image]
om/72776429/187825129-c3a69929-dd5f-461d-886f-7e9b048ae440.jpg)
cal, I got the back light changing
[image]
 the color. However it is unable to read the JPEG files.  

What can be the issue ?

Bodmer, 2022-09-01:
Do not connect anything to the SDO pin on the display. It does not go tristate so the SD card cannot be read.

More info here:
https://github.com/Bodmer/TFT_eSPI/discussions/898

closed 2022-09-01 by Bodmer

vick3394, 2022-09-03:
hello Mr Bodmer.
Thank you for reply.
I dont connect anything to the SDO because I dont use the touch screen.

I am doing just a simple slide show using the 4 inch TFT LCD Driver ILI9488.
The diode in the picture you included in the Github also does not exist.

When I run the TFT Meters, it works.
I used the Program that Xtronic did for the ILI9341
Just to read JPEG files from the SD card.

I know the program works because the background on the screen changes color as per program.
However it does not pick up the JPEG files in the SD card.

How should I proceed from here?

Looking forward to hear from you.

Regards
Victor

Bodmer/TFT_eSPI#898<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FBodmer%2FTFT_eSPI%2Fdiscussions%2F898&data=05%7C01%7C%7Cc19a89e2c4b44ed6c3d408da8c651cc0%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637976663191362509%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=nXkkKMJZBwFcb%2ByEmyKXfT6fNaGnqhUcfb4rAdXo03o%3D&reserved=0>