Bodmer/TJpg_Decoder #49: Problem With BYTECLIP Referencing version 0.2.0

Dragonwolf-Dominus, 2022-10-06:
Hi Bodmer.
I lost a big amount of time checking what was the problem with my code, and as I encapsulate all in objects, I thougth it was my fault, but now that I clear all warnings and bad practices from it, I see that I have the same error with BYTECLIP after compilation.
I tried to reference directly tjpgd.h and tjpgd.c, but it doesn't work and it's not a good thing anyway.
Searching the problem on the net, I finally found this thread, and of course, I effectively have the same problem. Platform IO only see the 0.2.0 version, and the olders ones (0.1.0 and 0.0.3) but no more. Nor the 1.0.2 or the 1.0.5.
Trying to force it with bodmer/TJpg_Decoder@^1.0.5 in platform IO, I have the same response ;
"Could not find the package with 'bodmer/TJpg_Decoder @ ^1.0.5' requirements for your system 'windows_amd64'"

You seem to make a great work, I have tested TFT-eSPI, and it's faster than Arduino GFX or Adafruit GFX, but I can't use it now as it needs TJPEG_Decoder to work as I want and I must try to put the last version to see if it resolves the problem with BYTECLIP.

I'm a bit disapointed by this problem... why it doesn't fail on compiling since it says a var is not defined.... why after at link time...
I'd like to understand, but what is sure, is that it comes from your package as it desapear since I remove it.

Thanks for your work. I hope I could see it working soon :)

_Originally posted by @Dragonwolf-Dominus in https://github.com/Bodmer/TJpg_Decoder/issues/41#issuecomment-1270131395_

Dragonwolf-Dominus, 2022-10-06:
I found a turnaround that works :
- Removing TJpg_Decoder from platform io project references.
- Downloadind the last version on git.
- Placing this version in the Lib folder of the project.
- Ensure deleting all other files from .pio/libdebs

It seem's to work for me, but I guess I won't have updates.

Bodmer, 2022-10-19:
For some reason some library updates do not get recognised, this seems to be a problem with the library scanner and not something I can fix.

closed 2022-10-19 by Bodmer