Bodmer/TJpg_Decoder #50: Compile error in All_SPIFFS example

hansaplasst, 2022-10-16:
Hi Bodmer,

I've tested TFT_eSPI and it works great! Super Nice work!
This time I'm trying to compile the All_SPIFFS example for my ESP8266 (NodeMCU clone) but I am getting this error.
```
/home/hansa/dev/Projects/Arduino/Projects/All_SPIFFS/All_SPIFFS.ino: In function 'void loadFile(const char*)':
All_SPIFFS:131:11: error: 'class TJpg_Decoder' has no member named 'getFsJpgSize'; did you mean 'getJpgSize'?
  131 |   TJpgDec.getFsJpgSize(&w, &h, name); // Note name preceded with "/"
      |           ^~~~~~~~~~~~
      |           getJpgSize
All_SPIFFS:140:11: error: 'class TJpg_Decoder' has no member named 'drawFsJpg'; did you mean 'drawJpg'?
  140 |   TJpgDec.drawFsJpg(0, 0, name);
      |           ^~~~~~~~~
      |           drawJpg
Multiple libraries were found for "SD.h"
 Used: /home/hansa/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SD
 Not used: /home/hansa/Installed/arduino-1.8.19/libraries/SD
exit status 1
```
Since it drawFsJpg and  getFsJpgSize are only loaded when TJPGD_LOAD_FFS is defined, I included this definition in the project.
But now when I am trying to compile it I get this error. 

```
/home/hansa/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: sketch/All_SPIFFS.ino.cpp.o: in function `_Z10tft_outputssttPt':
/home/hansa/dev/Projects/Arduino/Projects/All_SPIFFS/All_SPIFFS.ino:116: undefined reference to `_ZN12TJpg_Decoder12getFsJpgSizeEPtS0_PKcRN2fs2FSE'
/home/hansa/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/hansa/dev/Projects/Arduino/Projects/All_SPIFFS/All_SPIFFS.ino:116: undefined reference to `_ZN12TJpg_Decoder9drawFsJpgEiiPKcRN2fs2FSE'
/home/hansa/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: sketch/All_SPIFFS.ino.cpp.o: in function `_Z8loadFilePKc':
/home/hansa/dev/Projects/Arduino/Projects/All_SPIFFS/All_SPIFFS.ino:123: undefined reference to `_ZN12TJpg_Decoder12getFsJpgSizeEPtS0_PKcRN2fs2FSE'
/home/hansa/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /home/hansa/dev/Projects/Arduino/Projects/All_SPIFFS/All_SPIFFS.ino:135: undefined reference to `_ZN12TJpg_Decoder9drawFsJpgEiiPKcRN2fs2FSE'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
 Used: /home/hansa/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SD
 Not used: /home/hansa/Installed/arduino-1.8.19/libraries/SD
exit status 1

```
Maybe you can help this  hobbyist programmer out ;-)
Thanks!

-Hansa

Bodmer, 2022-10-19:
It looks like ESP8266 is not defined in newer board packages.

Comment out lines 1 and 3 in your User_Config.h file in the library, see here:
https://github.com/Bodmer/TJpg_Decoder/blob/master/src/User_Config.h

Bodmer, 2022-10-19:
This is now fixed in the master copy.

closed 2022-10-19 by Bodmer

hansaplasst, 2022-10-20:
Just checked and it compiled fine.
Thanks a lot!