Bodmer/TJpg_Decoder #53: Redraw TFT image if TFT SDO/MISO output is unavailable

hansaplasst, 2022-10-22:
Hi Bodmer,

I'm testing the Animated_dial example with an ESP8266 and [GC9A01 round display](https://nl.aliexpress.com/item/1005004342510301.html) without SDO. This means that with the existing code you cannot write image data over an existing background image without erasing (parts of) the data.

Without MISO however, we could use the original background image data from dial.h to but back clean image data to the TFT? Is this possible with your function library?

Bodmer, 2022-10-22:
Yes, it is possible and I have written an experimental sketch that works that way. I will see if I can find it.

hansaplasst, 2022-10-22:
Thanks for the quick reply! That would be awesome!

Bodmer, 2022-10-22:
It think I must have archived, it will be a couple of days before I can run a search.

Bodmer, 2022-10-22:
There is an example here:
https://github.com/Bodmer/TFT_eSPI/discussions/860#discussioncomment-208303

However you may find the esp8266 does not have sufficient RAM.

hansaplasst, 2022-10-22:
Thanks! I will check it out.

closed 2022-10-22 by hansaplasst

hansaplasst, 2022-11-03:
That example worked out great!
Here is what I've come up with.

[Animated_dial_noTFTrd.zip](https://github.com/Bodmer/TJpg_Decoder/files/9932935/Animated_dial_noTFTrd.zip)

Thanks again for your help!

hansaplasst, 2022-11-04:
BTW.   Does the pivot point need to be within a sprite's boundaries (width and height)?

Bodmer, 2022-11-04:
No, it can be outside. See notes [here](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Sprite/Rotated_Sprite_1/Rotated_Sprite_1.ino#L4-L6).

I tried your animated dial example and it works well and fast on an RP2040 with parallel display. Excellet!  I had some stray red pixels left by the tip of the needle.