Bodmer/TJpg_Decoder #54: Official RP2040 MBed support?

IAmOrion, 2022-10-23:
What's the chances this will eventually be able to support the Official RP2040 MBed boards instead of earlephilhower's ?

My projects uses USBHID which earlephilhower does not support, but I also want to use images in .h files using this TJpg_Decoder but apparently it doesn't work with the default RP2040 board selection

I'm buggered no matter what ha!

Bodmer, 2022-10-23:
The library has worked with the Arduino mbed RP2040 package in the past with the exception that LittleFS, DMA and PIO could be used. Some examples worked fine. I have not checked for compatibility recently since Phil's package works so well. I have only used keyboard USB HID with Phil's package though.

closed 2022-10-26 by Bodmer

Bodmer, 2022-10-26:
I have updated the library and tested the Flash_Jpg example with TFT_eSPI and the RP2040 Mbed board package 3.4.1. It compiles fine after a few tweaks, see below.

You will need to comment out line 2 in User_Config.h that loads FLASH filing system support, See here:
https://github.com/Bodmer/TJpg_Decoder/blob/master/src/User_Config.h#L2

You may also need to comment out lines thus //#include <pgmspace.h> in a sketch because the Mbed package does not seem to provide that header. It would be nice to automate these things but I cannot see a way of idenitfying which board package is being used... maybe there is a package signature macro that I have not noticed?

Bodmer, 2022-10-26:
I have made another change so User_Config.h does not need to be eddited.

IAmOrion, 2022-10-29:
Interesting... did you change anything else? I changed the user_config same as here: https://github.com/Bodmer/TJpg_Decoder/commit/41aaa5361a3a3e07730f46f27d13419e008a8a42

But I get an error as FS.h being missing

commit 83d86433 (2022-10-26): Fix #54

To use with the RP2040 Mbed Arduino board package the User_Config.h file in the src folder must be eddited to comment out line 2 thus:
```
  //#define TJPGD_LOAD_FFS
```
This is required since the Mbed package does not support the LittleFS or SPIFFS FLASH filing system. Clearly the LittleFS and SPIFFS examples will not compile for this reason.
commit 41aaa536 (2022-10-26): Fix #54