Bodmer/TJpg_Decoder #55: wrong decode of .jpg file

spilz87, 2022-11-21:
hello

I tryed the example of Flash_Jpg_GFX.ino from array, it works well.

but im trying to display my own image (download from a server)
[https://cdns-images.dzcdn.net/images/cover/8442e9ac0227a07b00c9dfd0ec00032d/50x50-000000-80-0-0.jpg](url)

i downloaded it.
I convert it through 
[http://tomeko.net/online_tools/file_to_hex.php?lang=en](url)

I get this :
`const uint8_t image[] PROGMEM = {
0xFF, 0xD8, 0xFF, 0xE0, 0x00, 0x10, 0x4A, 0x46, 0x49, 0x46, 0x00, 0x01, 0x01, 0x01, 0x00, 0x48, 
0x00, 0x48, 0x00, 0x00, 0xFF, 0xDB, 0x00, 0x43, 0x00, 0x06, 0x04, 0x05, 0x06, 0x05, 0x04, 0x06, 
0x06, 0x05, 0x06, 0x07, 0x07, 0x06, 0x08, 0x0A, 0x10, 0x0A, 0x0A, 0x09, 0x09, 0x0A, 0x14, 0x0E, 
0x0F, 0x0C, 0x10, 0x17, 0x14, 0x18, 0x18, 0x17, 0x14, 0x16, 0x16, 0x1A, 0x1D, 0x25, 0x1F, 0x1A, 
0x1B, 0x23, 0x1C, 0x16, 0x16, 0x20, 0x2C, 0x20, 0x23, 0x26, 0x27, 0x29, 0x2A, 0x29, 0x19, 0x1F, 
0x2D, 0x30, 0x2D, 0x28, 0x30, 0x25, 0x28, 0x29, 0x28, 0xFF, 0xDB, 0x00, 0x43, 0x01, 0x07, 0x07, 
0x07, 0x0A, 0x08, 0x0A, 0x13, 0x0A, 0x0A, 0x13, 0x28, 0x1A, 0x16, 0x1A, 0x28, 0x28, 0x28, 0x28, 
0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 
0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 
0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0x28, 0xFF, 0xC2, 
0x00, 0x11, 0x08, 0x00, 0x32, 0x00, 0x32, 0x03, 0x01, 0x11, 0x00, 0x02, 0x11, 0x01, 0x03, 0x11, 
0x01, 0xFF, 0xC4, 0x00, 0x1A, 0x00, 0x00, 0x02, 0x03, 0x01, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x04, 0x05, 0x01, 0x02, 0x06, 0x00, 0x03, 0xFF, 0xC4, 0x00, 
0x19, 0x01, 0x00, 0x02, 0x03, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x03, 0x04, 0x01, 0x02, 0x05, 0x00, 0xFF, 0xDA, 0x00, 0x0C, 0x03, 0x01, 0x00, 0x02, 
0x10, 0x03, 0x10, 0x00, 0x00, 0x01, 0x57, 0x9C, 0xF1, 0x31, 0x6B, 0x74, 0x98, 0xA3, 0x37, 0x64, 
0x2A, 0x0E, 0x0A, 0x75, 0xB4, 0x0C, 0xE6, 0xAA, 0x01, 0xA1, 0xD5, 0xA0, 0x27, 0x3F, 0x27, 0x51, 
0x43, 0xEA, 0x48, 0x89, 0xA9, 0x6B, 0x3D, 0x1A, 0x6F, 0x5D, 0xF4, 0xE4, 0xA3, 0x24, 0x25, 0x5C, 
0xA3, 0x4C, 0x53, 0x6F, 0x33, 0xAF, 0x8E, 0x75, 0x48, 0xE9, 0xE4, 0xCE, 0xBD, 0x04, 0x55, 0xB5, 
0x08, 0xE8, 0x91, 0x42, 0xE2, 0x1E, 0xCB, 0x32, 0xD5, 0xDB, 0x38, 0x99, 0xF3, 0x14, 0x03, 0x0B, 
0x72, 0x36, 0x7A, 0x2F, 0x95, 0x7B, 0x32, 0x26, 0x19, 0xB6, 0xB3, 0x82, 0x0E, 0x13, 0x70, 0x75, 
0x1C, 0x8B, 0x0F, 0x18, 0x4A, 0x15, 0x35, 0x96, 0x17, 0x18, 0x91, 0xA6, 0x54, 0xF6, 0x11, 0xFD, 
0xA3, 0xBF, 0xFF, 0xC4, 0x00, 0x22, 0x10, 0x00, 0x01, 0x04, 0x01, 0x04, 0x02, 0x03, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0x01, 0x02, 0x04, 0x11, 0x05, 0x10, 0x12, 
0x13, 0x14, 0x21, 0x31, 0x34, 0x42, 0xFF, 0xDA, 0x00, 0x08, 0x01, 0x01, 0x00, 0x01, 0x05, 0x02, 
0x88, 0xDB, 0x91, 0x9A, 0x03, 0x71, 0x75, 0xE7, 0xAD, 0x9E, 0x44, 0x68, 0x66, 0x35, 0xE3, 0x60, 
0x8F, 0x4D, 0xF3, 0x64, 0x2D, 0xE4, 0x5A, 0xE7, 0xDD, 0x29, 0x0E, 0x28, 0x8F, 0x82, 0xCB, 0x04, 
0x62, 0x60, 0x8F, 0xE2, 0x15, 0x58, 0xC4, 0x56, 0x5C, 0x88, 0x61, 0xF2, 0x21, 0xD1, 0x27, 0x5A, 
0x90, 0x9D, 0xAB, 0xD1, 0x66, 0x81, 0xBB, 0x62, 0x8F, 0x7B, 0x94, 0x34, 0xE2, 0xB4, 0xD0, 0xBE, 
0x7F, 0x57, 0x39, 0xCC, 0x63, 0x8A, 0xE1, 0x15, 0xD9, 0xC9, 0x02, 0xC4, 0xC0, 0x85, 0x97, 0x8C, 
0x25, 0xEE, 0x4F, 0x88, 0xBE, 0xC1, 0x00, 0xF0, 0x71, 0xC5, 0x56, 0xBA, 0xE2, 0x43, 0xD4, 0x2B, 
0x22, 0x58, 0x62, 0x0F, 0xB3, 0xD7, 0x24, 0x39, 0x3E, 0xD3, 0xD8, 0x1F, 0x5F, 0x6F, 0xFF, 0xC4, 
0x00, 0x24, 0x11, 0x00, 0x02, 0x02, 0x01, 0x03, 0x04, 0x02, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0x03, 0x21, 0x04, 0x11, 0x12, 0x10, 0x22, 0x31, 0x32, 0x13, 
0x14, 0x41, 0x61, 0x81, 0xFF, 0xDA, 0x00, 0x08, 0x01, 0x03, 0x01, 0x01, 0x3F, 0x01, 0x56, 0xC6, 
0x65, 0xBA, 0x95, 0x53, 0xC9, 0x56, 0xAE, 0x36, 0x3D, 0x91, 0xCD, 0x14, 0xCE, 0xC9, 0x36, 0x59, 
0x24, 0xE4, 0xB6, 0x3F, 0x8C, 0x8F, 0xB9, 0xAB, 0xF6, 0x44, 0x31, 0xBC, 0x91, 0x45, 0xAD, 0xF9, 
0x23, 0x95, 0xFA, 0x2C, 0x97, 0x36, 0x94, 0x4F, 0xAF, 0x31, 0x2C, 0x96, 0x42, 0x33, 0xEE, 0x65, 
0x92, 0x8D, 0x6F, 0x8A, 0x15, 0x8A, 0x2B, 0x72, 0xBB, 0xE7, 0x6F, 0x6B, 0xF0, 0x43, 0xB6, 0x47, 
0x32, 0x38, 0x3C, 0x8F, 0x3D, 0x34, 0xF1, 0x45, 0x5B, 0x45, 0xBD, 0xCE, 0x43, 0x1A, 0x3F, 0x3B, 
0x31, 0x9A, 0x65, 0xCE, 0xC4, 0x85, 0x14, 0xB0, 0x8E, 0x3D, 0x1A, 0x27, 0x1C, 0xE0, 0xE0, 0xCD, 
0x3C, 0x78, 0x77, 0x1F, 0x24, 0x99, 0x91, 0x13, 0x27, 0xD2, 0x8F, 0x5E, 0xBF, 0xFF, 0xC4, 0x00, 
0x25, 0x11, 0x00, 0x02, 0x02, 0x01, 0x03, 0x04, 0x01, 0x05, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
0x00, 0x00, 0x00, 0x01, 0x02, 0x03, 0x11, 0x10, 0x12, 0x21, 0x04, 0x13, 0x31, 0x32, 0x22, 0x14, 
0x41, 0x51, 0x52, 0xA1, 0xFF, 0xDA, 0x00, 0x08, 0x01, 0x02, 0x01, 0x01, 0x3F, 0x01, 0x96, 0xE8, 
0xF9, 0x29, 0xAE, 0x76, 0xAC, 0xC5, 0x97, 0x53, 0x64, 0x16, 0xE6, 0x6F, 0x69, 0x72, 0x59, 0x18, 
0xC2, 0x3B, 0xA2, 0x43, 0x2A, 0x2D, 0xBF, 0xE9, 0xB9, 0x7E, 0xD1, 0x25, 0xE8, 0xCE, 0x87, 0xD1, 
0x8F, 0x0F, 0x86, 0x5F, 0x5C, 0x65, 0xC0, 0xB3, 0x1F, 0x3E, 0xC5, 0x71, 0xD8, 0x9C, 0xA6, 0x7D, 
0x5D, 0x7F, 0x82, 0x5E, 0xA5, 0x6D, 0xD7, 0xF1, 0x45, 0x50, 0x94, 0x96, 0x64, 0xCE, 0xD3, 0x6C, 
0x7D, 0x35, 0x75, 0x3D, 0xD1, 0xF2, 0x75, 0x09, 0xCE, 0xBF, 0x07, 0x65, 0x8D, 0x64, 0xC2, 0x44, 
0x56, 0x16, 0x97, 0xB2, 0x72, 0xCA, 0xE3, 0x4C, 0x89, 0xE4, 0xFB, 0x08, 0xEA, 0x1E, 0xDA, 0xDB, 
0x32, 0xDF, 0x2C, 0xCE, 0x91, 0x65, 0x72, 0xCA, 0xE4, 0xDC, 0x8E, 0xA2, 0x7B, 0xBE, 0x26, 0xC4, 
0x63, 0x48, 0x95, 0x78, 0xD2, 0xFF, 0x00, 0x6D, 0x7F, 0xFF, 0xC4, 0x00, 0x29, 0x10, 0x00, 0x01, 
0x03, 0x03, 0x02, 0x05, 0x03, 0x05, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 
0x02, 0x11, 0x03, 0x12, 0x21, 0x31, 0x41, 0x10, 0x22, 0x32, 0x51, 0x71, 0x13, 0x42, 0x61, 0x52, 
0x62, 0x81, 0x91, 0x92, 0xFF, 0xDA, 0x00, 0x08, 0x01, 0x01, 0x00, 0x06, 0x3F, 0x02, 0xD4, 0x63, 
0x65, 0xCD, 0x4C, 0x99, 0xDC, 0x04, 0x05, 0x8F, 0x13, 0xDC, 0x2C, 0x10, 0x83, 0x5B, 0xF9, 0xF8, 
[... 57 lines cut ...]
0x32, 0x91, 0x02, 0x68, 0x28, 0x00, 0x68, 0x77, 0x53, 0x0A, 0x26, 0xB2, 0x80, 0xAD, 0xBB, 0x76, 
0xEF, 0x3D, 0xE1, 0x9A, 0x1F, 0x8B, 0x7B, 0x07, 0xCE, 0x3F, 0xAC, 0x93, 0xF0, 0x45, 0x22, 0x9A, 
0x63, 0x52, 0x7F, 0x79, 0xF7, 0x2B, 0xE7, 0x1B, 0xC4, 0x0A, 0x35, 0xD1, 0xD1, 0xBE, 0xBB, 0xF3, 
0x0A, 0xC1, 0x7B, 0x5B, 0x56, 0x0F, 0xE3, 0xB7, 0x93, 0xEF, 0x08, 0xC1, 0x75, 0x03, 0x53, 0x1C, 
0x3C, 0xE6, 0x58, 0xA2, 0x80, 0xF8, 0x17, 0x14, 0xF2, 0x45, 0x2C, 0xA3, 0xB5, 0xF8, 0x9E, 0x63, 
0x12, 0xEA, 0x83, 0x40, 0x3B, 0xFF, 0x00, 0x33, 0xEE, 0x60, 0x90, 0x2A, 0x84, 0xAE, 0x13, 0x50, 
0x48, 0x81, 0x0D, 0x8E, 0xFA, 0xEE, 0x33, 0xC3, 0x30, 0xBA, 0xBA, 0x21, 0xBD, 0xF1, 0xFE, 0x72, 
0xFF, 0x00, 0x0C, 0x0B, 0x29, 0xF3, 0x4E, 0xA6, 0x25, 0x32, 0xA8, 0x58, 0xCA, 0xCF, 0x49, 0xE1, 
0xE6, 0x09, 0xF9, 0xC8, 0x01, 0x74, 0xF8, 0xD7, 0xA6, 0x5F, 0xCE, 0x5F, 0xAD, 0x26, 0x04, 0xC0, 
0x7A, 0x7B, 0xC7, 0xC3, 0x39, 0xC1, 0x09, 0xD9, 0xB5, 0x3C, 0xD4, 0x99, 0xCA, 0xEB, 0x21, 0x0D, 
0x45, 0x4E, 0xBC, 0xE7, 0xDE, 0x04, 0x00, 0x28, 0x81, 0x48, 0x4A, 0x1B, 0x11, 0x7E, 0x2E, 0x75, 
0x18, 0x09, 0xA6, 0xDE, 0x3F, 0x38, 0xCD, 0x74, 0xBF, 0x7F, 0x5B, 0x9F, 0xAC, 0xA7, 0x6A, 0x2E, 
0x32, 0x84, 0x7C, 0x08, 0xC7, 0x28, 0x2E, 0xAB, 0x80, 0x4B, 0xF5, 0xFF, 0x00, 0xB8, 0x82, 0x80, 
0x4F, 0x8C, 0xFE, 0xDF, 0xFA, 0x60, 0x17, 0xA3, 0x20, 0x68, 0xE7, 0xC6, 0x7F, 0xFF, 0xD9};`

but when I tryed to use it in the example, after :
`TJpgDec.getJpgSize(&w, &h, image, sizeof(image));`
I got :
`Width = 0, height = 0`

it seems the decodage of the file is not working on this one, can anyone help me please :)

thanks in advance

Spilz

Bodmer, 2022-11-21:
Jpeg files can be in a number of different formats. The image is in "progressive" format, which is not supported. This format is typically used for large images, and initially decode as low resolution and the successively render with higher and higher resolution.

When the image is imported into a image handling program like MS Paint, then saved as a jpeg (default for Paint is to save in "Baseline" format) it renders fine.

closed 2022-11-21 by Bodmer

spilz87, 2022-11-21:
Thanks a lot for your very quick reply. 

As I try to load pictures directly from a website, I can't convert it. I need to find an other way to do 

Thank a lot and congrats for your job, everything is clear 👍👍👍