Bodmer/TJpg_Decoder #56: Cannot run on ESP32

Staccato-kashi, 2022-11-30:
I used someone's project written for ESP8266, and I'm trying to transplant it to my ESP32 based board on Arduino IDE, then I found that it could work on ESP8266 but not on ESP32-PICO. I've already sodered the chip on board… So is there anyway to make it enable on ESP32?

Bodmer, 2022-11-30:
The library has been tested with the ESP32.

Have you run some of the examples provided with the library?

Staccato-kashi, 2022-11-30:
> The library has been tested with the ESP32.
> 
> Have you run some of the examples provided with the library?

I tried "Flash_array" and "All_LittleFS", but nether of them works…or I should describe it more explicit, it cannot be verified. I think this because ESP32-PICO does not provides "LittleFS" library, but I don't know how to load it from another directory and compile it with new directory.
[image]

Bodmer, 2022-11-30:
The library needs to be used with a graphics library, which library are you using and do the examples for it work?

Bodmer, 2022-12-01:
I see from the screen shot that you are using TFT_eSPI, so run the Read_User_Setup sketch and verify it is as expected. Check other graphics examples work.

Check you are using the latest ESP32 Arduino board package 2.0.5,  older packages than 2.0.0 do not provide LittleFS.

Staccato-kashi, 2022-12-01:
Thanks for your help. It's my fault that I only tried to install ESP32 board package from _dl.espressif.com_ but not other way… I did't know the newest board package is on GitHub. I'm sorry to bother you about my careless mistake. (And my poor English, if so.)

closed 2022-12-01 by Staccato-kashi