Bodmer/TJpg_Decoder #58: Problem with specific SDcards

cobungra, 2022-12-10:
Hello, If you have a moment, I'd like your opinion.
I have had great success with TJPG_Decoder and TFT_eSPI (thanks!)
I recently bought a batch of SDcards which refuse to render the entire image on screen.
There is no problem reading the files. I can even send them from the SDcards and view the images via HTTP.
All of my other SDCards work fine...
I'm happy to bin these, but I just wonder what's going on?
(Using SD_Jpg Example / TJpg_Decoder 1.0.8)
Cheers.
Photos show the same image, same hardware and software but different brand of SDCard. (I have ten, same performance, ready for public disgrace...)

[image]
[image]

cobungra, 2022-12-12:
Not a problem for TJpg_Decoder. 
Thanks.

closed 2022-12-12 by cobungra

Bodmer, 2022-12-12:
Hi, glad to hear that. Thanks for reporting that.