Bodmer/TJpg_Decoder #59: Compile error ALL_LitteFS

hansaplasst, 2023-01-05:
Hi,

Had LittleFS working a while ago. But now for some reason it doesn't.
```
/home/.../Projects/All_LittleFS/All_LittleFS.ino: In function 'void loadFile(const char*)':
/home/.../Projects/All_LittleFS/All_LittleFS.ino:110:11: error: 'class TJpg_Decoder' has no member named 'getFsJpgSize'; did you mean 'getJpgSize'?
   TJpgDec.getFsJpgSize(&w, &h, name, LittleFS); // Note name preceded with "/"
           ^~~~~~~~~~~~
           getJpgSize
/home/.../Projects/All_LittleFS/All_LittleFS.ino:119:11: error: 'class TJpg_Decoder' has no member named 'drawFsJpg'; did you mean 'drawJpg'?
   TJpgDec.drawFsJpg(0, 0, name, LittleFS);
           ^~~~~~~~~
           drawJpg
Multiple libraries were found for "SD.h"
  Used: /home/.../.arduino15/packages/esp32/hardware/esp32/2.0.6/libraries/SD
  Not used: /home/.../.arduino15/libraries/SD
exit status 1

Compilation error: 'class TJpg_Decoder' has no member named 'getFsJpgSize'; did you mean 'getJpgSize'?
```
ESP32
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json

Bodmer, 2023-02-12:
I would drop back to IDE 1.8.19 or earlier as the latest 2.x.x IDE does not yet allow the extra tools to added upload the LittleFS files.

If the partition size is changed prior to a sketch compile and upload with IDE 2.x.x then it will be necessary to upload all the LittleFS files again.

closed 2023-02-12 by Bodmer

hansaplasst, 2023-02-12:
Linking in Arduino IDE 2.0.3 and 1.8.19 still fails :'( 

I've tried:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json (2.0.6)
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json (2.0.6)
and
https://dl.espressif.com/dl/package_esp32_index.json (2.0.6)

I have `tft_setup.h` in All_FS.ino folder and `platform.local.txt` in all directories which contain a `boards.txt`. I know this is deprecated but it works perfect here for TFT_eSPI. I could try changing `User_Setup.h` to match my `tft_setup.h`. Could this solve the error?

The only change I've made `All_LittleFS.ino` file is this:
```
#define ESP32
// Include the jpeg decoder library
#include <TJpg_Decoder.h>
#include "tft_setup.h"
```
If I don't include the `ESP32` definition `getFsJpgSize(...)` isn't found:
```
/home/cosmic/dev/Projects/Arduino/Projects/Examples/All_LittleFS/All_LittleFS.ino: In function 'void loadFile(const char*)':
All_LittleFS:111:11: error: 'class TJpg_Decoder' has no member named 'getFsJpgSize'; did you mean 'getJpgSize'?
   TJpgDec.getFsJpgSize(&w, &h, name, LittleFS); // Note name preceded with "/"
```
Seems logical since `TJPGD_LOAD_FFS` is defined with the `ESP32` definition...

I even tried different boards. But I am 90% sure it used to work with the "WEMOS D1 MINI ESP32" board.
Any idea what I could try next?

Thanks!

hansaplasst, 2023-02-12:
Here is my `tft_setup.h`:
```
#define ESP32
#define GC9A01_DRIVER

#define TFT_WIDTH 240
#define TFT_HEIGHT 240

#ifdef ESP32
  // WEMOS D1 MINI ESP32
  #define TFT_MOSI   23   // 23 DIN/SDA/SDI
  // #define TFT_MISO   -1   // SDO
  #define TFT_SCLK   18   // 18 CLK
  #define TFT_CS      5   // 5  CS/SS Chip select control pin
  #define TFT_DC     32   // Data Command control pin
  #define TFT_RST    22   // Reset pin (could connect to Arduino RESET pin)
  #define TFT_SDA_READ    // This option is for ESP32 ONLY
  #define ADC_RD     15   // Temperature pin
  #define ADC_RES  4095   // ADC Resolution
  #define TOUCH_RD    4   // Touch sensor pin (GPIO04)
  #define RSW_A      26   // Rotary switch A
  #define RSW_B      27   // Rotary switch B
  #define RSW_BTN    33   // Rotary switch Button
  #define BAUD_RATE 115200
#endif

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  66000000

#define SPI_READ_FREQUENCY  20000000 // 10000000 

#define SPI_TOUCH_FREQUENCY  2500000
```

Bodmer, 2023-02-13:
Trying to include the setup with the sketch causes all sorts of problems seemingly unrelated and these seem impossible to trace.

That is likely the root cause.

hansaplasst, 2023-02-13:
Totally my fault! 
Although I knew the previous method was broken I persisted in using it. Simply because it worked with TFT_eSPI.  :tired_face: 
Sorry for taking up your precious time.

I've implemented the [suggested method](https://github.com/Bodmer/TFT_eSPI#Tips). No errors anymore. It works!
Thank you!