Bodmer/TJpg_Decoder #61: JPEG Decode to BMP and Save to SD - Color Decoding Issue

RSViewMan, 2023-02-12:
Bodmer - Your library is awesome!! -- Thank you for publishing it.

I am attempting to save a decoded JPEG to BMP565 directly to SD and am having an issue with the color mapping not being correct.  Refer to the sample screen shots below.  The original JPEG is 1200x1600 and the Scale factor is 4.  I have tried different scales w/o any change in the output (other than size).  I have hard coded everything below to alleviate any possibility of a looping error or boundary miscalculation.  I thought maybe there was a byte misalignment in the file but the BMP Row Data ends on an even 4 byte boundary -- which appears to be a necessity of the BMP format.  The tjpgdcnf.h file is set for 565 Decoding (default setting).

The JPEG renders to a TFT display w/o any issues using your TFT Library -- the quality is great and the picture is clear.

**_Any input you could give regarding how to correct the BMP file conversion would be appreciated.  Thanks._**

[image]
[image]

````cpp
//======================================================================
#include <TJpg_Decoder.h>
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define HSPI_SCK  14
#define HSPI_MISO  12
#define HSPI_MOSI  13
#define HSPI_CS  15

SPIClass hSPI = SPIClass(HSPI);

// Support funtion prototypes
bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap);
void loadFile(const char *name);

#define FIRMWARE_VERSION "v0.0.1"

unsigned long currentMillis = millis();
unsigned long previousMillis = 0;  // will store last time LED was updated
const long interval = 10000;  // interval at which to blink (milliseconds)
bool firstScanComplete = false;
File root;
File file;
unsigned long tftRenderLoops = 0;
bool decodeToBMPFile = false;
File bmpFile;
//unsigned int bmparray[304][402];
unsigned char decoderRow[4][300*2]; //Hold rows for decoding
//unsigned char decoderRow1[612*2]; //Hold rows for decoding
//unsigned char decoderRow2[612*2]; //Hold rows for decoding
//unsigned char decoderRow3[612*2]; //Hold rows for decoding
//unsigned char decoderRow4[612*2]; //Hold rows for decoding
int16_t xPrevious=0;
String sourceFile = "/IMG_20230129_144628.jpg";
String destFile = "/test3.bmp";

// variables
bool shouldReboot = false;            // schedule a reboot

// function defaults
//String listFiles(bool ishtml = false);
//String testFunction(bool ishtml = false);

void setup() {
  Serial.begin(115200);

  Serial.print("Firmware: "); Serial.println(FIRMWARE_VERSION);

  Serial.println("Booting ...");

  Serial.println("Configuring HSPI Port");
  hSPI.begin(HSPI_SCK, HSPI_MISO, HSPI_MOSI, HSPI_CS);
  delay(2000);    

  Serial.println("==============================");
  Serial.println("SD Card Configuration Start");
[... 177 lines cut ...]
  // 16 = bits per pixel
  // all other header info = 0, including RI_RGB (no compr), DPI resolution

  unsigned long fileSize = 2ul * h * w + 54; // pix data + 54 byte hdr
  
  bmFlHdr[ 2] = (unsigned char)(fileSize      ); // all ints stored little-endian
  bmFlHdr[ 3] = (unsigned char)(fileSize >>  8); // i.e., LSB first
  bmFlHdr[ 4] = (unsigned char)(fileSize >> 16);
  bmFlHdr[ 5] = (unsigned char)(fileSize >> 24);

  bmInHdr[ 4] = (unsigned char)(       w      );
  bmInHdr[ 5] = (unsigned char)(       w >>  8);
  bmInHdr[ 6] = (unsigned char)(       w >> 16);
  bmInHdr[ 7] = (unsigned char)(       w >> 24);
  bmInHdr[ 8] = (unsigned char)(       h      );
  bmInHdr[ 9] = (unsigned char)(       h >>  8);
  bmInHdr[10] = (unsigned char)(       h >> 16);
  bmInHdr[11] = (unsigned char)(       h >> 24);

  bmpFile.write(bmFlHdr, sizeof(bmFlHdr));
  bmpFile.write(bmInHdr, sizeof(bmInHdr));
}

````
[image]

Bodmer, 2023-02-12:
I have not looked in detail at you sketch, but bad the output image suggests the image render program is correctly identifying the pixels as 16 bit and the pixel count is correct.  However the colours are either 1. not being interpretted correctly by the image display program or 2. pixels have not beem endcoded correctly.

Black is being correctly interpretted but this is a value of 0 to be expected.  I would check if your display program can handle 565 encoded pixels as it may be interpretting them as 4444 ARGB.  There online tools that will convert an image to 565 format so maybe test your image display program with an output file from that:
https://online-converting.com/image/convert2bmp/

If those steps do not reveal a problem then I would do a binary compare of a simple small image composed of solid red, green, blue, black and white blocks. You could then compare the output of your image converter with the output from the online tool.

RSViewMan, 2023-02-12:
Bodmer - Thanks for the quick reply.  I did a test using Red, Orange, Purple, Cyan and Blue JPEG files -- All tests created valid BMP files on the SD that accurately represented the source.  I will continue to look at the data output.  Thanks again.

closed 2023-02-12 by RSViewMan