Bodmer/TJpg_Decoder #63: JPEG rendered noisy, barely recognisable

boroncsokb, 2023-04-07:
Hi @Bodmer,

I am having the same issue as described in #34 . Everything worked great until I moved from Platformio to esp-idf (v4.4, using c++ and ESP32-S3 WROOM N4R8), but now I can not get it to work properly. The TFT_eSPI library still works, I can fill the screen with different colors, I can draw shapes on it without any problem, and also the TFT_Flash_Bitmap example works as intended, so I assume, that the communication with the screen is not the root cause.

However as I try to run the Flash_Jpg example from the TJpg_Decoder library (only switching the setup-loop functions to app_main for it to work in esp-idf), the image does not appear properly (see attached photo)
On the serial output, it also says: "Width = 240, height = 0"
I have already tried setting the SwapBypes to true and false, but I get the same noisy image every time.

Do you have an idea, how to fix this error?

Thank you!

[image]

Bodmer, 2023-04-07:
I have not used the esp-idf and have not seen this problem with the Arduino IDE.

Maybe you can get help on the Espressif forum.

mischief, 2023-05-02:
i've also got this problem. the JPEG renders fine with https://github.com/Bodmer/JPEGDecoder, but not with this library.

my hardware is `Seeed Studio XIAO ESP32C3` with an `ILI9341` display, using TFT_eSPI 2.5.23, and i'm building firmware with platformio.

notice the green and purple distortions in the render using TJpg_Decoder, when the original image is greyscale.

JPEGDecoder 1.8.1:
[image]

TJpg_Decoder 1.0.8:
[image]

original image file from SD Card ([source](https://en.wikipedia.org/wiki/File:Glenda_bunny_mascot_of_plan_9_from_bell_white.jpg)):
[image]

stripped down sketch:
```main.ino
#include <SPI.h>

#include <FS.h>
#include <SD.h>

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

#define TJPGD_LOAD_SD_LIBRARY
#include <TJpg_Decoder.h>

bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap)
{
   // Stop further decoding as image is running off bottom of screen
  if ( y >= tft.height() ) return 0;
  tft.pushImage(x, y, w, h, bitmap);  // Blocking, so only returns when image block is drawn
  // Return 1 to decode next block.
  return 1;
}

void setup() {
  Serial.begin(BAUD);

  // Set all chip selects high to avoid bus contention during initialisation of each peripheral
  digitalWrite(TFT_CS, HIGH); // TFT screen chip select
  digitalWrite(TOUCH_CS, HIGH); // Touch controller chip select (if used)
  digitalWrite(SD_CS, HIGH); // SD card chips select, must use GPIO 5 (ESP32 SS)

  while(!Serial) delay(100);

  if (!SD.begin(SD_CS)) {
    Serial.println("Card Mount Failed");
    return;
  }
  uint8_t cardType = SD.cardType();

  if (cardType == CARD_NONE) {
    Serial.println("No SD card attached");
    return;
  }

  Serial.print("SD Card Type: ");
  if (cardType == CARD_MMC) {
    Serial.println("MMC");
  } else if (cardType == CARD_SD) {
    Serial.println("SDSC");
  } else if (cardType == CARD_SDHC) {
    Serial.println("SDHC");
  } else {
    Serial.println("UNKNOWN");
  }

  uint64_t cardSize = SD.cardSize() / (1024 * 1024);
  Serial.printf("SD Card Size: %lluMB\n", cardSize);

  Serial.println("initialisation done.");

  tft.begin();

  TJpgDec.setCallback(tft_output);
}

void loop() {
  tft.setRotation(3);

  drawSdJpeg("/glenda6.jpg", 0, 0);
  delay(2000);
}

void drawSdJpeg(const char *filename, int xpos, int ypos)
{
  Serial.println("===========================");
  Serial.print("Drawing file: "); Serial.println(filename);
  Serial.println("===========================");

  uint16_t w, h;

  // Time recorded for test purposes
  uint32_t t = millis();
  JRESULT rv = TJpgDec.getSdJpgSize(&w, &h, filename);
  if(rv != JDR_OK){
	Serial.println("jpg error");
	delay(500);
	return;
  }

  Serial.print("Width = "); Serial.print(w); Serial.print(", height = "); Serial.println(h);

  TJpgDec.drawSdJpg(xpos, ypos, filename);

  t = millis() - t;
  Serial.print(t); Serial.println(" ms");
}
```

Bodmer, 2023-07-06:
Glad you have it working. Thanks for the example code.

closed 2023-07-06 by Bodmer