Bodmer/TJpg_Decoder #66: The image cannot be displayed in its entirety

pan1024, 2023-06-04:
Dear author I have encountered some problems, I am using ESP32C3 + ST775V2 240*320 + SD, JPEG decoding, I can use flash example for full image display, but when I use SD card can not be fully displayed, only about a third of the image is displayed, I switched the picture and SD card, also tried to use littlefs, but this problem is still not solved, And almost all images can only show the area on the picture, the upper left corner cannot be rendered, I use version 1.8.1
[https://img1.imgtp.com/2023/06/04/IttYq40X.jpg](url)

pan1024, 2023-06-04:
I just tested fetching pictures from the web and displaying them and it's no problem

pan1024, 2023-06-04:
Just now I tried to write the image in the SD card to littlefs and then decode it, and it was successful, so I think this problem may be caused by the operation of SD in the program

Bodmer, 2023-06-04:
Is the SD card is on the same SPI bus as the display? This can cause problems if tft.startWrite and tft.endWrite are not used correctly, because data access conflicts can occur in the SPI hardware buffer.

pan1024, 2023-06-04:
> SD 卡是否与显示器位于同一 SPI 总线上？如果未正确使用 tft.startWrite 和 tft.endWrite，这可能会导致问题，因为数据访问冲突可能发生在 SPI 硬件缓冲区中。

Yes they are in the same SPI, you mean to separate it when reading data is deactivated TFT?

pan1024, 2023-06-04:
> 

bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap)
{
  tft.startWrite();
   // Stop further decoding as image is running off bottom of screen
  if ( y >= tft.height() ) return 0;
  // This function will clip the image block rendering automatically at the TFT boundaries
  tft.pushImage(x, y, w, h, bitmap);
  tft.endWrite();
  // This might work instead if you adapt the sketch to use the Adafruit_GFX library
  // tft.drawRGBBitmap(x, y, bitmap, w, h);

  // Return 1 to decode next block
  return 1;
}

pan1024, 2023-06-04:
> bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap) { tft.startWrite(); // Stop further decoding as image is running off bottom of screen if ( y >= tft.height() ) return [...]

I did this but it doesn't seem to work

pan1024, 2023-06-04:
Also I found that when I use "TJpgDec.drawSdJpg(0, 0, "/panda.jpg"); , the SD card will not be able to read the data normally