Bodmer/TJpg_Decoder #67: Error compiling for Teensy 4.1

ZL1CVD, 2023-07-06:
Seems Teensy SD lib expects pFilename as const char* and not String..

error: cannot convert 'const String' to 'const char*'
  397 |   if ( !SD.exists(pFilename) )

Note I am not using any method that requires loading JPG from SD. My JPG is stored as const uint8_t logo[]  

Lib compiles & works fine for ESP8266, ESP32, ESP32-S3. Just has issues with Teensy 4.x

Bodmer, 2023-07-06:
As a work-around, comment out this line, recompile and upload:

https://github.com/Bodmer/TJpg_Decoder/blob/master/src/User_Config.h#L5

I have not tried the library with a Teensy so there may be other processor specific issues. If you do (or do not) get it working then post back.

ZL1CVD, 2023-07-06:
Worked a treat! Thank you so much! Now compiles without error. 
But note for ESP8266 I needed **TJpgDec.setSwapBytes(true);**
For Teensy 4.1 commented this out..

Thanks again for your assistance..

ZL1CVD, 2023-07-06:
Worked a treat! Thank you so much! Now compiles without error.
But note for ESP8266 I needed TJpgDec.setSwapBytes(true);
For Teensy 4.1 commented this out..

Thanks again for your assistance..

closed 2023-07-06 by ZL1CVD