Bodmer/TJpg_Decoder #69: "The file 'TJpgDec.h' could not be found."

peasdesign, 2023-10-17:
Dear Bodmer.

I am using your TJpg_Decoder library in my Arduino project. I am trying to decode a JPEG image from an SD card, but I am getting an error message that says "The file 'TJpgDec.h' could not be found."

I have checked the Arduino IDE library folder and I can see that the TJpgDec.h file is present. I have also checked the GitHub repository for your library and I can see that the TJpgDec.h file is present in the SRC directory.

I am wondering if there is a reason why I am getting this error message. Could you please help me to troubleshoot this issue?

I have provided the following information that may be helpful:

Arduino IDE version: 2.2.1
Board type: esp32 dev
TJpg_Decoder library version: 1.0..8
SD card image file name: /001.jpg
I would appreciate your help in resolving this issue.

Thank you for your time and consideration.

Sincerely,
P.D.

Bodmer, 2023-10-17:
You seem to have the wrong filename, it is TJpg_Decoder.h

peasdesign, 2023-10-18:
thank you for quick reply.
I am searching about 
TJpgDec.drawFsJpg()

I ll check it out again.

closed 2023-11-08 by Bodmer