Bodmer/TJpg_Decoder #70: Error Code 3 = JDR_MEM1

fah, 2023-11-07:
I get an Error Code 3 = JDR_MEM1,	/* 3: Insufficient memory pool for the image */

My code is:
```c
uint16_t w = 0, h = 0;
JRESULT resSize = TJpgDec.getFsJpgSize(&w, &h, temp_filename); 
if (resSize== JDR_OK) {
   ESP_LOGV(TAG,"Width = %i , height = %i \n",w,h);
} else {
   ESP_LOGE(TAG,"Size Error: %i\n", resSize);
}

JRESULT resDraw = TJpgDec.drawFsJpg(0, 0, temp_filename);
if (resDraw== JDR_OK) {
   ESP_LOGV(TAG,"JPEG drawn\n");
} else {
   ESP_LOGE(TAG,"Draw Error: %i\n", resDraw);
}
```
returns
```
Size Error: 3
Draw Error: 3
```

I checked that the file "temp_filename" in SPIFFS is valid with the FSWebbrowser example. 

Size of the File is 12148 Bytes. 

Dimensions: 704 x 576, 72 Pixel/Inch,Color model: RGB
JFIF: 1.0.1, Density 1/1, ColorSync: sRGB IEC61966-2.1    

First Hex values of the file: 
FFD8FFE0 00104A46 49460001

Controller is a 
LILYGO TTGO T5-4.7 Inch E-paper ESP32 S3 with 8MB 

The code, executed right before  TJpgDec.getFsJpgSize() and TJpgDec.drawFsJpg(), :
```c
ESP_LOGI(TAG, "RAM left: %d", esp_get_free_heap_size());
ESP_LOGI(TAG, "task stack: %d", uxTaskGetStackHighWaterMark(NULL));
ESP_LOGI(TAG, "Heap: Free: %i, Min: %i, Size: %i, Alloc: %i", ESP.getFreeHeap(), ESP.getMinFreeHeap(), ESP.getHeapSize(), ESP.getMaxAllocHeap());
```
returns

```
RAM left: 8290103
task stack: 5324
Heap: Free: 206380, Min: 202340, Size: 339580, Alloc: 196596
```
---

I am looking for a solution to display this image in 16 colors as greyscale on the above-mentioned e-paper display.
As far as I can tell, the size of the image is too large for the ESP. So a first step could be to reduce the 24-bit colours to 4-bit greyscale? Could I do that with another package that is compatible with yours?

Bodmer, 2023-11-08:
The library can decode large jpgs so I suspect there is something about the image file that casues a problem. Can you post the roguw image as a zip file here?

Bodmer, 2023-11-08:
Maybe if the image output is greyscale, this bug is the problem?
http://elm-chan.org/fsw/tjpgd/patches.html

Bodmer, 2023-11-08:
I have added the above patch.

fah, 2023-11-08:
Dear Bodmer, 

thank you for the premium support. :-)

I have applied the patch, but error 3 still persists.
Please find attached a tested zip jpeg file that causes this error.

[601_picture.jpg.zip](https://github.com/Bodmer/TJpg_Decoder/files/13293972/601_picture.jpg.zip)

Regards

Bodmer, 2023-11-08:
It appears the image format is not compatible with the TJpg_Decoder library decoder engine.  Since I did not write the core decoder I am not able to identify the root cause.

The image also crashes [this library](https://github.com/Bodmer/JPEGDecoder) which uses a different decode engine!

The [decoder library here](https://github.com/bitbank2/JPEGDEC) uses another decode engine and works with the troublesome image, so that would be a good option in this case.

Example use of JPEGDEC with TFT_eSPI:
[70_JPEGDEC_demo.zip](https://github.com/Bodmer/TJpg_Decoder/files/13300412/70_JPEGDEC_demo.zip)

closed 2023-11-08 by Bodmer

fah, 2023-11-08:
You are the best! Thank you very much!