Bodmer/TJpg_Decoder #76: Any way to fetch and display JPGs from the web without storing them to SPIFFS in between?

moeburn, 2024-03-20:
The JPG web fetch and display functions are great, but is it essential to store them on disk first, or can they be displayed straight from memory?  I want to use this function several hundred times a day, and that would wear out the flash memory in short order.

Bodmer, 2024-03-20:
Try the fork here and let me know if it works for you:
https://github.com/SirCooper/TJpg_Decoder

There are 2 streaming examples added in that fork.

kai11, 2024-05-05:
@Bodmer this streaming library works GREAT. Are you going to merge these changes?

RWBRWB, 2024-06-20:
Awesome streaming examples. Looking to download Album art for music and streaming directly to the LCD makes sense vs storing it.

nodoubtman, 2025-04-02:
You have to use // https://github.com/bitbank2/JPEGDEC.git
#include <JPEGDEC.h>

library... i uploaded a .rar file here:
[https://filebin.net/78n9x2zmk7coluke
](url)

hurry up because it expires in 6 days.

Hope it works for you.
It works for me flawlesslly.

Marc.