Bodmer/TJpg_Decoder #77: Unable to produce jpg that can be displayed

curious-emanuel, 2024-03-26:
I am unable to produce a custom jpg that would work. The library works with the provided panda image however that is the only image that works for me. 
I am trying to produce basic jpg images with gimp, attaching an example. It seems to have identical properties to the panda image. What could be going wrong?

Example image:
[image]

Blitz54, 2024-03-28:
When you export a jpg with GIMP, there is a drop down box for additional options. Open that box, and unselect "Progressive". If you're using the development build of GIMP, Progressive is near the top and not hidden.

curious-emanuel, 2024-03-30:
@Blitz54 You are the MVP. I had a feeling that I read somewhere that it **HAD** to be progressive. What a silly mistake. Thanks a lot!

ats3788, 2024-11-11:
I found a easy way Install Gimp with the addon https://github.com/alessandrofrancesconi/gimp-plugin-bimp/releases/tag/v2.6
then you remove Progessiv and as Baseline Thank you Hr. Bodmer for your Work